mortaca/RGB-Pi #2: Old French 50hz TV - Resolution issues

Dancorp, 2017-02-19:
Hi,
Some resolutions looks not being well supported be my TV. That's a vintage french one designed for a 50hz signal.

Photos : https://goo.gl/photos/M96FkKrf59NDq7EKA

On Recalbox Menu, the image is fullsized but cropped, and when i launch a game, I got a black frame on the top and the image is cropped / not centered
In retroarch setting, changing aspect ratio to 1:1 is better but i still got a black frame. (Refresh Fréq issue ?)

I looking for the service manual of the chassis of my TV to retune the geometry settings but can you share a way to test some resolution @ 50Hz ?

BR

Dan 
ps : sorry for my english, i'm french.

mortaca, 2017-02-20:
This is probably by the hdmi_timing selected, you can play with the timings if enter by the web service in setting section or editing recalbox.conf and you find all the resolutions for every emulator.

[Here](https://github.com/mortaca/RGB-Pi/blob/master/HDMI_Timings.txt) in the repository you can find some resolutions to try, in the next releae all the resolutions are most polished to evite this issues, don't worry not is a big problem to solve.

Sir-Ironic, 2017-02-20:
Hi Dancorp (I'm french too)
Your TV is very old school :)
Before attempting to change the settings of your TV, we can adjust the timings.
As morcata said, better timings coming.
You can contact me to adjust them.

Are you using 50Hz consoles?
Try this:
1920 1 140 200 260 256 1 18 10 29 0 0 0 50 0 39400000 1

Dancorp, 2017-02-20:
Thanks for your response 
I have try your timing but this one is worse !
According to mortaca suggestion, i will try some others but in any case, i have to wait for the next update.
When did you have planned to release it ?

Sir-Ironic, 2017-02-20:
I think new release is coming. Probably few days.
You want timing for what ?
For 224p/240p at 50Hz ?
As mortaca said, have a look to timings list

Dancorp, 2017-02-21:
Thanks Sir-Ironic but using your settings (the first and the second one) that's what i got : 
https://goo.gl/photos/yKvX7ihVKDQniNzX6
🤒

Sir-Ironic, 2017-02-21:
It's that. It's a resluton for emulatrors.
You probably want a low resolution for EmulationStation ?

Dancorp, 2017-02-23:
I'm looking for a working and user friendly resolution for both. It doen't matter if it is not the native resolution for each emulator as long as it keep playable.

Sir-Ironic, 2017-02-23:
ES : 
hdmi_timings 450 1 50 30 85 288 1 10 1 25 0 0 0 60 0 9600000 1

50Hz : 
hdmi_timings 1920 1 48 192 240 288 1 6 3 16 0 0 0 50 0 37560000 1

Dancorp, 2017-02-23:
Ok my friend, i just understand there are two timings to set.
One for ES by editing the config file at startup but where put timings for emulators ?

mortaca, 2017-02-25:
At recalbox.conf, you can edit this file by the web system.

Sir-Ironic, 2017-02-25:
I worked on several full screen (no overscan) for consoles.
All this timings are tested and work very great.
As you see, i tried to reach true refresh rate, not only 50Hz/60Hz
If you want only 60Hz timings, tell me.

You can try this and tell me if they are overscan or not.
My only problem is to handle 50Hz/60Hz NES games. Resolutions are ok but 
emulaators don't swith to correct PAL/NTSC format.

# NEOGEO
neogeo="hdmi_timings 1920 1 152 247 280 224 1 13 8 19 0 0 0 59.186 0 
40610000 1"

# PCENGINE
pcengine="hdmi_timings 1920 1 158 250 286 232 1 4 3 23 0 0 0 59.94 0 
41060000 1"

#PLAYSTATION
playstation60="hdmi_timings 1920 1 152 247 280 240 1 1 7 14 0 0 0 60 0 
40860000 1"
playstation50="hdmi_timings 1920 1 132 247 280 288 1 3 3 18 0 0 0 50 0 
40240000 1"

# MEGADRIVE
megadrive60="hdmi_timings 1920 1 48 192 240 240 1 3 3 16 0 0 0 59.92 0 
37680000 1"
megadrive50="hdmi_timings 1920 1 48 192 240 288 1 6 3 16 0 0 0 49.70 0 
37340000 1"

# MASTER SYSTEM
mastersystem60="hdmi_timings 1920 1 48 192 240 240 1 3 3 16 0 0 0 59.92 
0 37680000 1"
mastersystem50="hdmi_timings 1920 1 48 192 240 288 1 6 3 16 0 0 0 49.70 
0 37340000 1"

# NES
nes60="hdmi_timings 1920 1 158 200 286 240 1 1 8 13 0 0 0 60.10 0 
40380000 1"
nes50="hdmi_timings 1920 1 158 200 314 240 1 23 6 43 0 0 0 50.01 0 
40450000 1"

# SNES
snes60="hdmi_timings 1920 1 160 200 286 224 1 9 8 21 0 0 0 60.10 0 
40410000 1"
snes50="hdmi_timings 1920 1 158 200 314 239 1 25 6 42 0 0 0 50.01 0 
40450000 1"

*-- don't take care about /Estimated Monitor Frame/, it was not sync 
when i grab it --*

I work a lot to produce very few data but with this timings, i have a 
fullscreen (1920x240) screen area.
Look my RetroArch parameters, true full screen with 1920x240 viewport.

(224p is near fullscreen and 240p has a little oversan. I can't show 
more than 236 lines on my TV in 60Hz)

--------------------------------

What do you want to support with databases, only MAME/FBA games or all 
consoles games too ?
There is no games resolution databes, we must creat them, it is a 
really, really lot of work.

Nice weekend...

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

Dancorp, 2017-03-01:
Now upgraded to Raspberry 3, i have try to update Recalbox to 4.0.2 and now there is not more black frame on the top (on emulators) but the image is still cropped.
Neogeo games are less cropped but they still a little bit laggy. 
Changing recalbox conf file is confusing, i can't see any effect on my issue.
Something must be wrong, but despite my will on tuning Retroarch, i have not enough knowledge to solve it by myself. 
I will wait for the new Rgb-Pi update, that's official !
(I dream of the possibility to calibrate screen like in kodi.)

Sir-Ironic, 2017-03-06:
There are two way to calibrate monitor.
1) Use monitor/TV setting (menus/potentiometers).
1) Use hdmi_timings (it's more difficult).
Perfect automatique screen calibration isn't possible due to not equal screen settings form all TV/CRT.

mortaca, 2017-03-07:
You can not update recalbox by the official system, that erases the necessary changes for rgb-pi

Dancorp, 2017-03-07:
_Perfect automatique screen calibration isn't possible due to not equal screen settings form all TV/CRT._
i agree with you, but kodi allows to tune your screen size.
I understand it could be close to impossible to do that in Recalbox.
I love my vintage TV but it has some lacks for tuning the screen size. lacks of variable resistors and of course no osd service mode

Sir-Ironic, 2017-03-08:
You can calibrate screen with retroarch menu (Viewport).
But the goal is to reach perfect pixels. So, resize picture isn't the solution.
Please, explain me (accurate) again all you problems to see what i can do for you.

Dancorp, 2017-03-11:
aTg has given me the chance to try the alpha version of the next beta release of RGB-Pi
Thanks to new timings, (i guess), image is not cropped anymore in emulators.
I still have horizontal black frames due to the 60hz refresh rates not natively supported by my TV.
That's not a big issues in my opinion.
The only main problem, just reported to aTg, is the kodi timings.
Unlike prevous version, i got now the black frame and of course, i'm unable to remove it using the kodi screen calibration tool.
I have quickly recorded a short video to show you my results.
https://www.youtube.com/watch?v=MmutEDONNXI

Sir-Ironic, 2017-03-12:
I think Kodi is launched in 60Hz.
It's better to launch it in 288p 50Hz to have a full screen.

Dancorp, 2017-03-13:
I have found the way to have a full screen kodi (overscanned but calibrated) by adding a dedicated timings to it. (the same as ES)
kodi.videomode=hdmi_timings 480 1 14 45 56 300 1 10 5 5 0 0 0 60 0 9600000 1

using the global.videomode, i can easily test timing and this 50Hz one is the best for the moment (tested on snes).
hdmi_timings 1920 1 100 203 303 256 1 14 14 28 0 0 0 50 0 39400000 1

https://goo.gl/photos/GijRxvu6f1Sh7TN27

Dancorp, 2017-03-13:
@Sir-Ironic 
En français ca srait bien plus facile de discuter je pense :)
J'ai testé pas mal de tes timings 50Hz, mais les emulateurs ne se lancent pas ou l'image est complètement décalée vers le haut sur mon écran

Mon mail dbattistelli@gmail.com 

Ce message d'autodétruira dans 5s....

Sir-Ironic, 2017-03-14:
Perfect.
Just modify the pixel clock 39400000 to 39410000
At 39400000 your refresh rate is 49.992Hz.

Dancorp, 2017-03-14:
Thanks i will try this pixel clock.
Did you have any ressources to understand  each parameters on timings, and how to calculate them ?
I would like to find by myself a way to increase the vertical size to have a true fullscreen image on my screen.

Sir-Ironic, 2017-03-14:
I use C.R.U to calculate timings.
https://www.monitortests.com/forum/Thread-Custom-Resolution-Utility-CRU
Use the detailled resolution Windows
https://www.blurbusters.com/wp-content/uploads/2013/02/toastyx.jpeg
Respect : 
- Horizontal Khz of 15.5 (+ or - 0.3)
- Pixel clock more than 38.4Mhz
- Total lines up to 262 in 60Hz and 312 in 50Hz
- Horizontal front porch + sync width = back porch (but can be different).
- Vertical sync 3 to 10.
Change value, one by one and try.
Read online informations about timings and porchs/sync.

Dancorp, 2017-03-14:
That's exactly what i was looking for ! Thanks a lot :)

mortaca, 2017-07-20:
Now at the last pre-release you can edit the files to center the screeen from the TV. Writeme  to the mail to give the las release. I'm going to close this issue.

closed 2017-07-20 by mortaca