mortaca/RGB-Pi #4: Problems with visualization os the frontend

mortaca, 2017-03-07:
There are a number of monitors that are giving problems to display the frontend of emulationstation.

This is because you use a resolution of 480x300 to get a higher resolution, some TVs do not support more than 288 or 240 lines high so the following problems may arise.

     -Black image
     -Desynchronization
     -See the darkest image

This has very easy solution changing the resolution and will be solved in the next update but for those who need an immediate solution can do the following.

Edit the config.txt file of the boot partition, and replace the following line;
hdmi_timings=480 1 14 45 56 300 1 10 5 5 0 0 0 60 0 9600000 1

By this;
hdmi_timings=320 1 12 32 44 240 1 6 10 6 0 0 0 60 0 6400000 1

Edit the file recalbox.conf and replace the following line;
system.es.videomode=hdmi_timings 480 1 14 45 56 300 1 10 5 5 0 0 0 60 0 9600000 1

By this;
system.es.videomode=hdmi_timings 320 1 12 32 44 240 1 6 10 6 0 0 0 60 0 6400000 1

Reboot.

yusay, 2017-03-08:
Everything ok, i've solved the problem with this. TY very much

riper, 2017-03-12:
Hi.
I had that desynchronization issue and I tried to use that workaround. It works at the first boot, but when I play any MAME or PiFBA game and I get back to the menu, the resolution changes and I get again that problem.
I can send some pictures to show that effect.
It only happens getting back from MAME or FBA to frontend.
Everything's fins on Megadrive, Snes and so on.
Thanks.

mortaca, 2017-03-14:
@riper Thanks for warning, there was an error in the line of code, test in system.es.videomode without the symbol =

riper, 2017-03-14:
I've tried to access to the web interface (tried on firefox, chrome and ie) and I'm not able to enter.
Tried using the name or the IP (http://recalbox/     or     http://192.168.1.ip/ )  (also with and without the slash (/))
Maybe it's disabled on this version? I'm doing something wrong?

Sir-Ironic, 2017-03-19:
This is a link to google sheet.
https://docs.google.com/spreadsheets/d/1vH6JEG0_Xw2YNiBsknx-wX-ayCCsVdAlYyNCfVkKp2E/pubhtml

You can find lists of

ALL YOKO : Big list of honrizontal arcade games.
ALL TATE : Big list of vertical arcade games.
MAME YOKO : My yoko mame games.
MAME TATE : My tate mame games.
FBA YOKO : My yoko fba games.
FBA TATE : My tate mame games.
TIMINGS : An exhaustive list of Timings i use (not all are tested).

I used a lot of timings.cfg, too much timings.
I know you want to use just a few timings.

My config (database) file (for example mame yoko games) is :
(space is tabulation)
aburner2    yoko    224    5964    After Burner II
actfancr    yoko    240    6000    Act-Fancer Cybernetick Hyper Weapon 
(World revision 2)
airwolf    yoko    224    6000    Airwolf
aliens    yoko    224    6000    Aliens (World set 1)
arkarea    yoko    192    5961    Ark Area
...

A script translate this to array (really quick to read).
mame_games[aburner2.zip]='yoko224p5964'
mame_games[actfancr.zip]='yoko240p6000'
mame_games[airwolf.zip]='yoko224p6000'
mame_games[aliens.zip]='yoko224p6000'
mame_games[arkarea.zip]='yoko192p5961'
...

When a mame game is launcher, if this game is in database (if not, a 
video say game is launched in defaut resolution)
A resolution is launched (let's take aliens.zip) :

Timings array is loaded (. /recalbox/share/RGB-Pi/HDMI_Timings.cfg)
Game database is loaded (. 
/recalbox/share/RGB-Pi/Resolutions/mame_yoko_array_games.cfg)

yoko224p6000 is read from database array.

vcgencmd is launcher from r224p6000variable.
conf="${mame_games[$filename]}"
conf=`echo "r"${conf:4}`
eval "vcgencmd ""$""$conf"

And retroarch libretro mame is launched with yoko224p6000.cfg additional 
configfile.
retroarch -L $retroarchcores/mame078_libretro.so --config 
$retroarchconfigs/retroarchcustom.cfg --appendconfig 
$retroarchconfigs/${mame_games[$filename]}.cfg "$1"

I know, i know, this is not what you want to do, but it can be 
interesting for you to look at this.

You can download my last release (just for me, no published) here :
Link coming soon...

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

mikoyv3, 2017-03-20:
I have the same problem, when I play any game and go back to the menu, the resolution changes and I get again that problem. 

@mortaca can you please list all the different locations you changed? thanks.

riper, 2017-03-20:
@mikoyv3 , If you're using the Alpha version of the distribution you have to know that that's a very early (and so far not finishet) version. 
In any case, I've found the solution (in my case) to fix our problem. You can try to apply it on your pi.
I'm not in front of the computer so the names maybe are not 100% correct:
- Find a file called **allresolutions.conf** or something like that. You can find it accessing to the Pi by \\\RECALBOX  (and then on **RGBPi** I think)
- Open that file and search for something called **EMULATIONSTATION**, and there you will find another "system.es.videomode" instance. Change it by the one that mortaca says and that's it.
It worked for me.

But remember, that's an early build of the final system.

mikoyv3, 2017-03-22:
I think the system.es.videomode hdmi_timings only takes effect once you exit out of a game. I get a weird behaviour when on boot the screen has overscan.

1. Mount the USB in windows
/recovery/config.txt
dtoverlay=vga666
enable_dpi_lcd=1
display_default_lcd=1
#dpi_output_format=6
dpi_group=2
dpi_mode=87
hdmi_timings=320 1 12 32 44 240 1 6 10 6 0 0 0 60 0 6400000 1

2. out sd in usb and edit using linux
/root/recalbox/share_init/system/recalbox.conf
system.es.videomode=hdmi_timings 320 1 12 32 44 240 1 6 10 6 0 0 0 60 0 6400000 1

3. Open the pi's ip address using a browser
/recalbox/share/system/recalbox.conf
system.es.videomode=hdmi_timings 320 1 12 32 44 240 1 6 10 6 0 0 0 60 0 6400000 1

riper, 2017-03-23:
Have you tried what I said on my previous post?

mortaca, 2017-07-20:
This issue is solved by the last release, now only need to put the hdmi_timing on the config.txt at the first line  and the scripts take this line to restore the resolution when you exit from the game. I'm going to close.

closed 2017-07-20 by mortaca