mortaca/RGB-Pi #5: No image in Samsung CB-20F32Z

mortaca, 2017-03-08:
This TV model does not display picture under any resolution.

I am waiting to do the test powering with 5v pin 8 of the scart to check if this particular model needs to activate switching to show image.

Otherwise, it will be incompatible.

Sir-Ironic, 2017-03-08:
I have 4 TV CRT and the only one which does not work is the samsung.It a 14" (36cm) SAMSUNG TF-14N3

I have sound only.

mortaca, 2018-02-23:
The user tuskador of the forum of EOL has tested on his samsung to feed pin 8 of the scart with 8.8v (I imagine that with 12v will also work) and the problem has been solved.
I am already working on an additional stepup that can be installed inside the scart housing.

mortaca, 2018-03-02:
I'll leave it open until the new cable redesign is launched.

mortaca, 2019-08-13:
Nes cable design have a 12v on pin 8 to make it compatible with SAMSUNG TVs

closed 2019-08-13 by mortaca