mortaca/RGB-Pi #6: [4.1 Alpha2] Unable to launch Kodi

Dancorp, 2017-05-25:
Hi aTg,
Thanks a lot for this new release. 
I use my RGBPied TV to play games but also to stream old videos/film.
In the 4.1 Alpha, when i try to launch kodi, i got the recalbox splashscreen for 1 second and i return to ES.
Thanks in advance for your help. (I had to downgrade to Beta 3 for the moment)

Dan

mortaca, 2017-07-20:
No have any resolution at the moment to view Kodi in good conditions, for example ithe DVD resolution, by this reason don't work Kodi at this moment, but I leave open this issue to solve in the future.

webodan, 2017-11-29:
Has any progress been made on this? Would be much better to use rgb-pi with kodi than an og xbox or a PC botched together running kodi on top. Specially for SD content in RGB.

mortaca, 2017-11-29:
I have make some test and the problem is with Kodi options, te minimun resolution selectable are 1280x720 and the most hight resolution of the TV are 1280x288, a lot of vertical lines are lose, if someone found the way to launch Kodi at low resolution, I can include it by default on RGB-Pi OS

Dancorp, 2018-03-30:
Hi Mortaca,
Given that it seems very complex to make kodi work again, is it possible to modify LibreElec or OpenElec for the RGB-Pi cable?

mortaca, 2018-03-30:
The problem not is the OS, Kodi needs 720p resolution to work, is the lowest res configurable for kodi and 4/3 TVs of 50/60Hz only show 480i and by the moment with RGB-Pi only is possible show 240p or 288p.
You can run Kodi but lose a lot of lines.

Dancorp, 2018-03-30:
Your first Recalbox distrib was able to launch Kodi in 240p. It was Kodi 15.2 (Isengard).
I'm trying to figure out why it worked before...

mortaca, 2018-03-30:
I try in the Kodi of the Recalbox releases and i can't but you can run OpenELEC with RGB-Pi wollowing the steps of the FAQ to view other systems and try to force Kodi to 240p.

Dancorp, 2018-04-01:
Thanks mortaca.
I finally managed to launch LibreElec with a good resolution using your FAQ and timings of your first distribution.
hdmi_timings=480 1 14 45 56 300 1 10 5 5 0 0 0 60 0 9600000 1
I also found a great skin here : https://github.com/YggdrasiI/skin.confluence.480

**Still a problem,** I can't get sound through the GPIO. Configuring Kodi with the audio setting "Analogue"  does not change anything.

mortaca, 2018-04-01:
Great! You can make public the distribution of your mod of LibreElec for RGB-Pi I think a lot of people can enjoy of your work.

For the sound try installing the addon "audio profiles" and set the analogue output from this addon.
Another options is try to force the output writing this line on terminal:

sudo amixer cset numid=3 1

If you finally think in share your work with the comunity I can publish your project on the official twitter account of rgb-pi and link on the web to redirect all the users to your soft.

Dancorp, 2018-04-01:
Thanks for your support!
Unfortunately, neither solution works. Isn't the problem on the dtbi?

I admit that creating a distribution from my changes is not easy for me. I'll find out a little bit about that.
In the meantime, I can do a tutorial to explain step by step what to do.
But before that, the sound has to work first!

mortaca, 2018-04-01:
Review if you have pwm-2chan.dtb or .dtbo overlay on your /boot/overlays folder.

Dancorp, 2018-04-01:
pwm-2chan.dtbo is included.

mortaca, 2018-04-01:
Try this two options on config.txt

hdmi_ignore_edid_audio=1
hdmi_force_edid_audio=1

If don't work change the values between 1 and 0 in the 4 posible combinations.

mortaca, 2018-05-06:
I've been doing a lot of testing, changing the Kodi skin and reducing the scale but there is no way, the resolution is anchored to 720p and nothing can be done inside Recalbox, when it is for HDMI the menu have the option to change resolution until 640p but once you boot from the GPIO port the resolution options disappear and it automatically goes to 720p.

Whoever wants to use Kodi can install it with OpenELEC and configure it for RGB-Pi.

closed 2018-05-06 by mortaca

Dancorp, 2018-07-02:
Hello there!
I would like to try this command to solve my libreelec sound problem:  dtoverlay=pwm-2chan,pin=18,func=2,pin2=13,func2= 4

But I have to know which gpio pins are used for audio output on your cable :)

Thanks in advance for your help.

Br

Dan

Dancorp, 2019-04-23:
Hi Mortaca.
FYI, with the last LibreElec  9.x, the sound problem is fixed.
The last issue is about screen calibration settings but i work on it.
https://github.com/xbmc/xbmc/issues/15165

Will your MongOS distrib allow to watch video files?

BR

Dan

DaVukovic, 2019-04-24:
@Dancorp 

>The last issue is about screen calibration settings but i work on it.

The screen calibration in Kodi is fixed. As you are using LibreELEC 9.0.1, the version you have installed simply doesn't include the fix. LibreELEC 9.0.1 ships Kodi 18.1. The fix is included in Kodi 18.2 but there's currently no LibreELEC version which ships that Kodi version. Kodi 18.2 will be shipped with LibreELEC 9.0.2 once we released it ;)

For the time being you could test one of the milhouse builds: https://forum.kodi.tv/showthread.php?tid=298461

That will contain a newer Kodi version, but expect issues with those builds. Most things are working but there's not every corner-case covered.

kocane, 2021-07-31:
@Dancorp did you ever figure something out with Kodi?

Dancorp, 2021-07-31:
I must admit that I put my research on hold
I tried to find a solution with the French Facebook community but nobody is interested.
The video game is very good, but I do not understand why developers do not offer a video player, even through a dedicated distribution
It doesn't seem complex to me...

kocane, 2021-07-31:
> I must admit that I put my research on hold
> I tried to find a solution with the French Facebook community but nobody is interested.
> The video game is very good, but I do not understand why developers do not offer a video player, even through a dedicated distribution
> It doesn't seem complex to me...

Agreed. Very odd.

kocane, 2021-08-01:
@Dancorp Have you tried the CRTPi project though? Comes with Kodi I think

rtomasa, 2021-08-01:
Please avoid using this closed issue as a forum or chat. Recalbox distro was deprecated more than 2 years ago. The current OS includes a simple videoplayer.