mortaca/RGB-Pi #7: [4.1 Alpha2] Controllers unresponsive in-game

Coestar, 2017-06-08:
I experienced this issue on a clean installation of 4.1 Alpha2.

I have tried the following controllers: iBuffalo Classic USB Gamepad, Hyperkin Pixel Art Controller, and two 8BitDo NES30 Pro controllers.  All of them work correctly in Emulation Station menus.  Once I load a game, they are no longer responsive and/or seem to be mapped incorrectly.

I used "Super Mario Bros." on NES60 and "Super Mario All-Stars + Super Mario World" on SNES60 for testing.  The iBuffalo and Hyperkin controllers are both SNES-like, so they only map to D-PAD, A, B, X, Y, START, SELECT, L and R (I assigned "hotkey" to L).  In a game, none of the buttons do anything, and special mappings such as START+SELECT (to exit) don't work either.

The 8BitDo controllers are a little different.  They feature a more modern button layout, including all of the above buttons plus L2, R2, L3, R3, and two thumbsticks.  With these controllers, I mapped "hotkey" to L2.  Again, in a game, none of the mappings seem to work as they should.  The D-PAD is unresponsive.  The left stick works for movement instead.  B button works, but doesn't seem to map to NES/SNES B button (incorrectly does a "spin" jump in Super Mario World).  A, X, Y, L1, and R1 do nothing.  L2 (hotkey) and R2 act as SELECT and START respectively, though they were not mapped as such, and can be pressed simultaneously to exit a game.

All of these controllers work correctly in menus and games on RetroPie, and the iBuffalo controller worked correctly in menus and games on RGB-Pi 4.0 Beta3.

I apologize if this is a simple mistake on my part and not a bug.

Sir-Ironic, 2017-06-08:
Hi.
Sorry, the Alpha was an Alpha ;/
Next (soon) alpha will have correct mapping.

With my NES30 PRO 8bitdo and Recalbox 4.1beta (no RGB-Pi mod), the thumb is mapped for move, not the cross.

Next Alpha is coming...

Coestar, 2017-06-08:
Understood.  I wasn't sure if it was a known issue, nor if it is bug reports are desired.  I will patiently await the next update!

closed 2017-06-08 by Coestar