mortaca/RGB-Pi #9: Fix for namco/taito games in mame 0.78 core

mdd45, 2017-09-02:
Fix for Namco/Taito games in mame 0.78 core (Splatterhouse, pacman etc)

mortaca, 2017-09-03:
What exactly is the problem? I have tried pacman without problems.

mdd45, 2017-09-03:
please try these roms: 
mspacman.zip, splatter.zip, galaga88.zip . 

Either not boot at all (or randomly boot 1/30 times) or just black screen.

mdd45, 2017-09-03:
Please check this: https://github.com/libretro/mame2003-libretro/issues/22

mortaca, 2017-09-03:
But I can't compile mame in Recalbox, this is work of the Recalbox developers not mine.

mdd45, 2017-09-04:
I think here they say that recalbox team already pushed a fix: https://forum.recalbox.com/topic/2348/mame-splatterhouse-anyone-get-to-work/31

Maybe Ironic knows something more as he followed this topic in the past. Else implementation of AdvMame or mame4all core could be a "temporary" fix for us.

Sir-Ironic, 2017-09-04:
I tried.
Splatter House works but graphics a broken (splatter.zip).
MS. Pac-Man doesn't work, back to ES (mspacman.zip).
Galaga "88 doesn't work, back to ES (galaga88.zip).

And yes, advmame can be used to play this games, i tried and they all work perfectlly

But a feature of RGB-Pi is the ability of use other cores.
In Resolutions/mame078_games.txt you can see each rom use a core named mame078_libretro.so.
It's the last version of Mame78.
But you can use older version of Mame78 or iMame4All.

I tried ;
Splatter house with iMame4All and it works nice.
Ms Pac-Man with an older version of Mame078 and it works nice.
Galaga '88 with iMame4All and it works nice.

http://rnc.free.fr/RaspberryPi/libretro/imame4all_libretro.so
http://rnc.free.fr/RaspberryPi/libretro/mame078old_libretro.so

I must verify something, i think i forgotten the ability to choose the core on RGB-Pi.
A script must be modified :)

mdd45, 2017-09-04:
So should we expect the next version of rgb-os or there is a tutoria on how to use those cores? Thank you Ironic and Atg for your efforts. Rgb-os is a great distro!!!

Sir-Ironic, 2017-09-04:
Ok. To use any cores this the latest Alpha, you must :

- Put the libretro libraries (.so) in /usr/lib/libretro/
(iMame4All and Mame078old) See links above.

- In /recalbox/share/RGB-Pi/Emu_Launcher.py replace
**commandline = "retroarch -L %s/mame078_libretro.so --config %s --appendconfig /tmp/retroarcharcade.cfg \"%s\"" % (libretro_path,configgen_retroarchcustom,rom_full_path)**
by
**commandline = "retroarch -L %s/%s --config %s --appendconfig /tmp/retroarcharcade.cfg \"%s\"" % (libretro_path,games_datas[game][12],configgen_retroarchcustom,rom_full_path)**
MAKE A FILE BACKUP IF YOU DON'T KNOW PYTHON.

- In /recalbox/share/RGB-Pi/Resolutions/mame078_games.txt replace
**mspacman 1920 224 60.606060 -8 -25 5 48 192 240 5 15734 mame078_libretro.so V**
by
**mspacman 1920 224 60.606060 -8 -25 5 48 192 240 5 15734 mame078old_libretro.so V**

**splatter 1920 224 60.000000 -8 -25 5 48 192 240 5 15734 mame078_libretro.so H**
by
**splatter 1920 224 60.000000 -8 -25 5 48 192 240 5 15734 imame4all_libretro.so H**

**galaga88 1920 224 60.000000 -8 -25 5 48 192 240 5 15734 mame078_libretro.so V**
by
**galaga88 1920 224 60.000000 -8 -25 5 48 192 240 5 15734 imame4all_libretro.so V**

I just do it and all three games are working.
Sorry, this feature was brocken, i forget to try it :/

You can do the same for FBA and user cores .30, .34, .36, .37, .39.....
Some game run better this core .34, .36

closed 2017-09-04 by Sir-Ironic

mdd45, 2017-09-04:
Thank you i will try it asap

mdd45, 2017-09-04:
Cannot copy files inside RECALBOX\root\usr\lib\libretro due to permissions. Tried through network transfer and winscp. Any help?

Sir-Ironic, 2017-09-04:
If you want to enable write on Recalbox from Windows in one click, you can.
Use this executable + batch :
http://rnc.free.fr/RaspberryPi/scripts/plink/PLINK.EXE
http://rnc.free.fr/RaspberryPi/scripts/plink/RW_Recalbox.bat

mdd45, 2017-09-04:
had to edit your Emu_Launcher lines but got it to work.

so replace this: 

commandline = "retroarch -L %s/mame078_libretro.so --config %s --appendconfig /tmp/retroarcharcade.cfg \"%s\"" % (libretro_path,configgen_retroarchcustom,rom_full_path)

with:

commandline = "retroarch -L %s/%s --config %s --appendconfig /tmp/retroarcharcade.cfg \"%s\"" % (libretro_path,games_datas[game][12],configgen_retroarchcustom,rom_full_path)

only two backslashes were missing. Thank you. 

Now i can play SplatterHouse on RGB-PI ;-)