mortaca/RGB-Pi #10: Sound issue with cable

Drasicbcn, 2017-09-04:
Hi everyone,
I have a sound issue with the cable. It's like a noise that doesn't stop.
In the attached file it starts at 00:08 sg and it keeps going.
I've tried it on my 3 CRT and I have the same issue.
Thanks.

[Audio_170831_2.zip](https://github.com/mortaca/RGB-Pi/files/1274512/Audio_170831_2.zip)

mortaca, 2017-09-04:
This occurs because the audio is digital and filtered, some frequencies escape to the low pass filter that has RGB-Pi installed.
This noise is only perceptible at very high volume and when no audio is played.
It seems that it coincides with the reading and writing of data, we are seeing this causes other interferences also in the image, a possible solution is to reduce the processes running in the background.
We are looking for the solution, a temporary solution is to use the minijack port audio that is best filtered.

Sir-Ironic, 2017-09-04:
You can try this, i don't know if it really work.
Add this two lines in config.txt.
**
audio_pwm_mode=2
disable_audio_dither=1
**
And try to disable Wifi if you use it, prefer ethernet cable.

Drasicbcn, 2017-09-04:
Oh thank you very much!!
Disabling Wifi solved the problem at 95%, there's a little little background noise.
That two lines only remove sound from the games.

closed 2017-12-06 by mortaca