mortaca/RGB-Pi #11: No game launch

Jul1oS, 2017-09-13:
Hi,

I have The RGB PI with the latest firmware (4.1 Alpha3 RPi3), an RPI3 and a little 15inch [TV Sony Trinitron KV-M1450B](https://cdn.wallapop.com/shnm-portlet/images?pictureId=45486311&pictureSize=W320).
No games are launched, with every launch I come back to the menu Recalbox.

Why ?
Thanks

Sir-Ironic, 2017-09-13:
Hi.
Can you tell us more ?
- What Gamepad/Joystick do you use ?
- Do you use any other USB device ?
- What is your complete configuration ?
Thanks

Jul1oS, 2017-09-13:
My complete configuration

- RPI 3
- RGB PI 4.1 Alpha3 RPi3
- [TV Sony Trinitron KV-M1450B](http://img.abrakaba.com/00397D3B-0/Televiseur-Sony-Trinitron-KV-M1450B-35cm.jpg)
- Bluetooth [keyboard Rapoo](http://www.hkepc.com/database/images/2011/11/source/2211394181634760725.jpg) for test
- Only change horizontal resolution in Python Screen Utility in 320px for the TV

Thanks

mdd45, 2017-09-13:
maybe its a problem with gamelist.xml files in rom folders?

Jul1oS, 2017-09-13:
gamelist.xml is on every rom folder with the true names, i don't think the problem is here

Jul1oS, 2017-09-13:
i think its a problem with the 1920px horizontal definition and my crt tv but 
I do not know what to fix or how to

Jul1oS, 2017-09-13:
neogeo become great but the other i have nothing : black screen and return to recalbox menu

Sir-Ironic, 2017-09-14:
_Only change horizontal resolution in Python Screen Utility in 320px for the TV_
You must not set **1.HORIZONTAL SIZE** to 320.

**1.HORIZONTAL SIZE** isn't the real horizontal number of pixel, it's only a zoom parameter.The resolution is always 1920 horizontaly.

- Put back the **1.HORIZONTAL SIZE** to 0
Try 5 and -5 to see the diffence.
(-5) must shrink the game display to fit the screen.
(5) must enlarge the screen if you have black border.

Jul1oS, 2017-09-14:
All values ​​are zero except screen resolution which is at 320 otherwise it does not work on my tv. No games even pass those preinstalled in the system I always return to the menu except for the neogeo. There is a problem with the emulator can be? Or emulator settings? Or the timings.cfg file?

Sir-Ironic, 2017-09-14:
If you return to menu (emulationstation) it could be an open screen error

Could you post the contents of files (text files):
/recalbox/share/system/logs/es_launch_stderr.log
/recalbox/share/RGB-Pi/su.cfg
/tmp/logs.rnc

(try to launch a game before).

Did you try to reinstall RGB-Pi ?

Jul1oS, 2017-09-14:
It works thank you

I see a log in this file who a joystick device must be plugged in for launch games
/recalbox/share/system/logs/es_launch_stderr.log

Only a bluetooth keyboard doesnt launch game and the log say "No joystick device number found"
I plug a Joystick standard USB xbox 360 generic and all the games launch :)

Thank you very much

Sir-Ironic, 2017-09-14:
Happy for you ;)
You help me, i'm coding a joystick utility and i haven't think to the possibility of no joystick plugged ;)
Good games.

closed 2017-09-14 by Jul1oS