mortaca/RGB-Pi #12: The color red is not displayed properly on sony trinitron.

Karmeck, 2017-11-05:
http://imageshack.com/a/img922/8863/X8SdgF.jpg

In the image above I present to you the color red in the 240p test suite. This is on a Sony trinitron kv-29cl11e, Scart 1.   Scart 2 however give almost no image but sound. 

I have been told this is because of the output resolution of the rgb-pi of 1920.

Any input on this?

Sir-Ironic, 2017-11-06:
I must try other resolution but it seems to be a SONY TRINITRON issue only.
And only TRINITRON bigger than 51 cm seems to be affected.
For this time, i must get feedback on this issue.

Karmeck, 2017-11-06:
I'm about to buy a 25 inch trinitron later this week. Will report back. It's a Sony Trinitron KV-X2503E

mortaca, 2017-11-08:
I think this can be produced by a specific front porch values or pixelclocks of hdmi_timings

mikechi2, 2017-11-10:
Hi, this is Mike Chi, the RetroTINK guy. Per Karmeck's request, I'm posting some suggestions. I got the image working on both a VGA666 and my RGB24 hat. Aside from my American TV (Sony KV-27FV310) not syncing to the 50Hz menu output, everything works fine. 

For SNES, I would recommend:

hdmi_timings=2048 1 180 202 300 240 1 3 5 14 0 0 0 60 0 42954545 1 #240p - NTSC perfect clock based on SNES/NES timings

This is the exact timing from the hardware scaled by a factor of 8. The 1920 timings do not appear to match NTSC spec, and is likely the culprit.

mortaca, 2017-11-12:
Try changing this lines on /shar/RGB-Pi/Fonctions.cfg

	# Calculate of the Vertical Front Porch.
	#V_FP=V_Total-V_Res
	#V_FP=V_FP-V_Sync
	#V_FP=V_FP/2
	# Round (down) the Vertical Front Porch.
	#V_FP=int(floor(V_FP))
	

	# Vertical Position MUSTN'T be bigger than Vertical Front Porch.
	#if V_Pos > V_FP :
	#  V_Pos = V_FP
  
	# Calculate Vertical Front Porch.
	#V_FP=V_FP-V_Pos
	V_FP=1

Coment all lines and add to the end V_FP=1, in the timing for Emulationstation the vertical front porch are 1 and vertical sync too, with this changes put all timings with vertical front porch to 1, let me know the results.

Karmeck, 2017-11-12:
Tested the above code.

V_FP=1
Nothing changed. Red still not red. 

The new hdmi timings from mikechi2 put inside config.txt
Emulationstation now show the same bad red as inside emulators.

mortaca, 2018-01-22:
A possible solution provided from the shmups forum

https://shmups.system11.org/viewtopic.php?f=6&t=61114

With the original control of the tv sony you must enter the service mode, (turn off the TV to the standby mode, and press: info, 5, volume +, display) the TV will turn on automatically in the service mode, and you can see large green letters, there, you must press 34 on the remote.

Karmeck, 2018-01-22:
Oh, you found that. Was gonna include that in my goodbye note. Yes I'm
saying goodbye.

I have jumped ship to retrotink. (and I kinda regret it)

My rgb-pi is sold and I'm sending it today.

Thank you for all the help. You and I are alike, difrens is you have the
technical know how to actually do somthing. But we both help out fellow
people.

I will continue to spred the word of rgb-pi. (you could however make it
easier and actually include the price on the web site) and it should be
"at" not "in".
You can ask about availability in

Once again, thank you.

mortaca, 2018-02-21:
The user danytyler of the EOL forum may have found the solution to this problem.
It would be an insufficient ground problem in the GPIO pins, the temporary solution consists of bridging the pin 6 and 8 or if you prefer to open the scart and lengthen the cable that corresponds to pin 8 to the pin 6 wire.

If it is confirmed that this is the solution, I will review the PCB to use another additional ground cable.

I await your reports.

Karmeck, 2018-02-21:
Did you totally miss that I have sold my rgb-pi? I'm using vga and crt
monitor now.

I can not test this.

mortaca, 2018-02-21:
It is not a message addressed to you, this problem is still open for other users until it is solved.

mortaca, 2018-03-02:
Ok, It seems that the enigma of the red color is already solved.
Apparently some newer trinitron models are especially sensitive to a too strong RGB signal, as the signal in RGB-Pi is generated within the scart so close to the input and has no attenuation this affects these particular monitors.

The solution is to add attenuation to the signal, by means of resistance to 120 Ohms, can be added in a scart cable extender, in the next cable revision this will be included by default.

I keep the topic open until the new revision is launched.

mortaca, 2018-04-04:
Well, I have finally found the definitive solution to this problem without the need for any hardware modification, the problem was a pulse in the horizontal syncronism too high, this in some TVs generates a flattened color scale, can be corrected by lowering the "H Pulse" field in the hdmi_timings, it is possible that by lowering it timings with frequencies below 60Hz, they will suffer vertical desynchronization, so the "V sync" would also have to be lowered.

All the hdmi_timings will be reviewed in a following update and I will solve the problem.

mortaca, 2018-05-06:
This issue will be completely resolved in the next update in the next few days.

closed 2018-05-06 by mortaca