mortaca/RGB-Pi #13: somes bugs

dawiid77, 2017-12-11:
hello,
is it possible to fix some bugs please ?
sommes games are slow on snes ;
advance mame loose good controls after 1st round or stage or randomly ;
can’t setup rgb pi for a old flat CTR screen for no distortion ;
32x games 60hz have horizontal latence (pitfall us) ;
sega CD won't work despite correct bios for recalbox

thanks for your job !!

David
Regards

mortaca, 2018-01-13:
Your are mixing different things, most of this problems are not problems from RGB-Pi are issues from  Recalbox an need to be solved en her repository or forums.

About the problem with the old TV you need to be more specific, what model of TV you have and some pictures of the issue, in the menu and inside the games.

advmame is imposible to run inside retroarch, is a different emulator.

dawiid77, 2018-01-13:
For the ctr tv, it’s a flat screen.
I have distorsions bécasse the screen is flat. I incrise zoom to have a good apparence but I miss some 2 cm all over the screen.

Ok for advmame. Just say me how to remap controllers : because advmame arcade game  are no well mapped on automatic mode..

Thanks

mortaca, 2018-01-13:
But RGB-Pi is not designed for flat screens, are only for CRT screens.

dawiid77, 2018-01-13:
It’s bot a lcd flat screen.
It’s a CRT FLAT SCREEN Samsung relased on 2005.

How can I remap controls on advmame please ?

mortaca, 2018-01-13:
The file with the config of advmame is /share/system/.advmame/advmame.rc

The issue with the TV are only in the menu or inside the games too?
You try inside ScreenUtility to change systen resolution from 270 to 240?

dawiid77, 2018-01-13:
Thanks

Look at my tv please

When I put the zoom to 10

mortaca, 2018-05-06:
I have reviewed all the consoles for the next update.

closed 2018-05-06 by mortaca