mortaca/RGB-Pi #14: Netplay Possible?

webodan, 2018-01-13:
I've attempted arranging a netplay game with a friend and some cores crash when going into netplay -> Start hosting, and with the ones that do work, we can't type in the IP of either host, as there's no on-screen keyboard nor we can type it using the controller. Is this feature even working in this distro? I thought so, since it's based on RetroArch which has had its netplay code overhauled recently with very impressive results.

mortaca, 2018-01-13:
This is job for Recalbox team, I don't know how it works on the default RB distribution but you need to sear on RB documentation and forums, I don't touch nothing about this to adapt the system to CRT screens.

webodan, 2018-01-13:
Reading up on it, it says that recalbox 4.1 does include a retroarch version that is netplay compatible. Maybe this fork is using a slightly older version?

mortaca, 2018-01-13:
About the keyboard on the screen the OSD messages are disabled because with the resolutions used by the system the messages are unredeable.

You need to search first on the RB forums about this feature and if you can play with the default RB over hdmi online after I can found the issue on RGB-Pi but first is necessary discart one problem with RB.

closed 2018-01-20 by mortaca

mortaca, 2018-05-06:
I play Netplay from retroarch without problems.