mortaca/RGB-Pi #24: [Bug] Frequency Mode selection screen: input swap between P1 and P2

modology, 2018-07-01:
Please video for more details
https://youtu.be/6cwvii_3EP4

At frequency selection screen, the input from P1 is automatically swapped to P2 therefore make it impossible for player to continue to choose the mode without resorting to using controller 2

Suggested solution: allow any controllers (p1,p2,p3,p4,etc...) to select frequency mode

mortaca, 2018-07-01:
The selector is setup for "js0" the first usb attached to the pi, if you change the second joystick in Recalbox to control the player one this don't affect to the selector. Is better respect the order of the usb's because not all the emulators recognise the config from recalbox.

modology, 2018-07-01:
cant you just allow any controllers to make a selection at the selector screen? that way both P1 and P2 can choose if they wish to

mortaca, 2018-07-01:
I can look at it for a next update.

mortaca, 2018-07-26:
You have a temporary option to solve this, you can assign the selector to player 2 by easily modifying a field in the file /share/RGB-Pi/emu_launcher.py set this option to 1 to control the selector with player 2

`	joystick = 0 # 0 is the 1sf joystick`

YanikTheYak, 2018-12-14:
I modified the code to not display the selector, and instead use 60hz. It now loads the fba libretro emulator and works correctly. 
I think the joypad mechanism must not like the ipac?

mortaca, 2019-08-13:
Solved on the new OS

closed 2019-08-13 by mortaca