mortaca/RGB-Pi #30: Not supported on Raspberry Pi 4

squidrpi, 2019-07-06:
Plug the cable in, copy the rgb-pi.dtbo to /boot/overlays and add settings in config.txt (same as RPi3) but no image appears on the CRT.

Is this a software issue or hardware incompatibility?

rtomasa, 2019-07-08:
There is something you are doing wrong because it has been already tested in the latest Raspbian Buster and  it was working fine.

squidrpi, 2019-07-08:
On Buster with Raspberry Pi 4?
Followed the instructions in https://github.com/mortaca/RGB-Pi/blob/master/README.md but I get a blank screen.

rtomasa, 2019-07-08:
Yes Buster is the only supported for Pi4. Following the instructions also and worked fine. If still not working, try commenting out the 2 lines in the [pi4] section of the cfg related to the new video driver but do note that I was able to make It work without this change.

squidrpi, 2019-07-08:
Yes tried all that. Doesn't work on Pi3 either with this Buster image. I'll wait a bit for this to all mature. Don't want to keep unplugging the cable incase I break it.

squidrpi, 2019-07-09:
Tried with a new Buster image and added the normal rgp-pi to config.txt and it now works on the Pi4! It must have been because it was an early Buster image or something got messed up on that installation.

closed 2019-07-09 by squidrpi