mortaca/RGB-Pi #44: Amiga not still not working after Update on 2019.08.06

hardmanm, 2019-08-12:
game show up after latest update (ADF files) but the emulator does not boot up and games.  I did check that I have the right bios files installed which I did, but still not working.

mortaca, 2019-08-13:
Te emulator only load the .uae files, on next update we will remove the extensions .adf and .hdf so that they are not visible

closed 2019-08-13 by mortaca

hardmanm, 2019-08-14:
I believe that LR-PUAE should support multiple filetypes:-

Using lr-puae for Amiga emulation

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Using lr-puae for Amiga emulation

I just spent a few evenings getting this to work. Here's a bit of what I learned. Not being an original Amiga us...
 |

 |

 |

 
 
Closed #44.