mortaca/RGB-Pi #45: Neo Geo Sound Stuttering and MSX not working at all

cherryuk, 2019-08-13:
The sound stutters in all Neo Geo games. All roms tested on other devices, Retropie, PI2SCART all work without sound stuttering.

MSX roms do not work at all. Confirmed same roms working on other devices.

mortaca, 2019-08-13:
Maybe you have a bad configuration in your EQ if you have touched this option that may be causing distortions or a temperature too high in your CPU, neogeo works without any distortion in the sound. The MSX theme cannot mix two things in the same thread.

closed 2019-08-13 by mortaca

cherryuk, 2019-08-15:
After a lot of testing with Flat Sound EQ, checking temp of RPi. The Neo Geo sound is perfect only in FBA Arcade folder/emulator, same ROM, the Neo Geo folder/emulator has sound issues, any way to change the settings to match the FBA Arcade? Both use 1.7.5 – FB Alpha v0.2.97.44 cfbe1e1

mortaca, 2019-08-15:
Check your bios on neogeo folder, is the only thing different, the core are the same.