mortaca/RGB-Pi #52: Older versions?

squidrpi, 2020-05-23:
Tried OS v2 but I have big problems with it. Where are the previous versions?

rtomasa, 2020-05-23:
Please, instead of requesting old non supported versions, you could explain your issues so that we can try to fix them in the current OS.

squidrpi, 2020-05-23:
RGB OS v1 worked perfect for me, guess this is the problem with closed source. By mistake I installed the v2 Beta without keeping the previous version. My iBuffalo gamepad doesn't work. Can define keys but it won't work in any game (keyboard works fine). Also when I unplug the gamepad and replug it in I get an error and the gamepad won't work, I have to reboot.

rtomasa, 2020-05-23:
Your complain has no sense because both v1 and v2 are closed source created by me and both share the same code for controller manager, also there are more people using such pads, so sounds weird to me. Are you using RGB-Pi cable? Which version of the OS did you test? v2 BETA 1, 2, 3, 3.1, 3.2...?

You must provide full detailed setup and steps when opening an issue otherwise we won't be able to fix anything is there is anything really broken.

squidrpi, 2020-05-23:
RGB-Pi cable with OS v2 BETA 3.2, RPI 3B+.

rtomasa, 2020-05-23:
Please add some screenshot of the error (drag & drop of the image in the textbox). Also if you can attach the rgbpi.log file from /home/pi/RGB-Pi/logs it would be helpful

squidrpi, 2020-05-23:
Only logs are rgbpi.log:
[2020-05-23 17:05:14] Error in Init Joysticks: Invalid joystick device number
Traceback (most recent call last):
  File "input_manager.py", line 337, in init_joysticks
pygame.error: Invalid joystick device number

[2020-05-23 17:06:40] Error in Init Joysticks: Invalid joystick device number
Traceback (most recent call last):
  File "input_manager.py", line 337, in init_joysticks
pygame.error: Invalid joystick device number

Input error hasn't occurred on the screen the last few disconnects.

I still can't use the iBuffalo in any game, works fine on the front end.

rtomasa, 2020-05-23:
Games rely on RetroArch, it has nothing to do with my frontend, if it does not work it is probably because you have a keyboard or something conflicting with the pad, or because there is not any RetroArch autoconfig file generated.

1. Try removing any keyboard if plugged in
2. Check if there is any file in `/opt/retropie/configs/all/retroarch/autoconfig` If there is not any file just do the mapping again to force its generation

squidrpi, 2020-05-23:
gamepad works fine if I unplug the keyboard. Gamepad + keyboard worked without problems in OS v1.

rtomasa, 2020-05-23:
Just before closing the issue, I want to explain you that the controller subsystem is the same in both v1 and v2 (it is one of the few parts that have not been fully rewritten from scratch). That said, there are two different scenarios here:
1. The core/s that have been updated now prioritises the KB over the gamepad (I don't think that happened and also I cannot do anything here).
2. And this is the most probably scenario, you have the KB plugged into a USB port with priority over the one of the gamepad. Do note that RaspBerry USB ports have a sorting priority depending on the port number, that is, looking at the USB ports, they are internally numbered in U (top left (1), bottom left (2), bottom right (3), top right(4)) so if you want to keep your KB plugged for something special you should always use latest one, that is, port 4.

closed 2020-05-23 by rtomasa