mortaca/RGB-Pi #61: How can I add this arcade game's background manually?

ng0757, 2021-05-24:
I found this game pgalvip(fbneo) has no image to show. 
I checked in scraper_75.csv,
This game does not exist in this CSV file. How can I add it manually?

rtomasa, 2021-05-24:
You can simply add your image to /RetroPie/images/arcade folder with the name of the rom device in png format

ng0757, 2021-05-24:
But I've done it.
sftp://192.168.8.102//home/pi/RetroPie/images/arcade/lr-fbneo/pgalvip.png
sftp://192.168.8.102//home/pi/RetroPie/images/arcade/lr-mame2003/dino.png
...
...
But there was still no my-background image, so I had to follow the scraper_75.csvto cover the images in the /RGB-Pi/images/scraper/75/ directory,
I think there must be some problems that prevent the correct display of the pictures. The pictures I put in the Retropie/images/arcade/lr-fbneo/ directory can't be displayed. 
I have set the backgrounds option in the system to custom.

rtomasa, 2021-05-24:
The arcade image folder is shared among arcade cores so put your image in Retropie/images/arcade instead of Retropie/images/arcade/lr-fbneo/

ng0757, 2021-05-24:
Still can't show the bg image, and every arcade game can't show my bg image.
But NES game can display my bg image, even my video can have my own bg image, just put the bg image in the corresponding directory.

What's wrong with arcade bg ? I'm confused.

rtomasa, 2021-05-24:
Please for additional support go to the official Discord channel

closed 2021-05-24 by rtomasa