mortaca/RGB-Pi #62: It's not bugs,just 2 questions.hope to be solved :)

ng0757, 2021-05-29:
It's not bugs,just 2 questions.
1:
Is there any way to automatically load a game every time I start rgb-pi? For example, I like STG games. I want to automatically load ket.zip every time I turn on the power. When I press hotkey + B + B to exit, it will return to my favorites list.
2:
When I set all the settings "perfectly", how can I disable the function of pressing the start key to switch to the options panel, so that no one can change the settings unless I change them manually.

rtomasa, 2021-05-29:
1. Use autoplay option from game list options menu
2. Use kiosk mode in system options

closed 2021-05-29 by rtomasa

ng0757, 2021-05-29:
Thank you for your reply. I have ignored these options before:)