mortaca/RGB-Pi #63: Calculate new timings

digitalLumberjack, 2021-07-12:
Hello i'm trying to set new timings to fit some other consoles on Recalbox.

Could you provide me some informations about how to create them ?

Are there some rules I must follow in order to have good timings ?

Thank you.

mortaca, 2021-07-12:
I suppose you mean the Pi3 because there is a lot of difference between 3 and 4 due to the peculiarity that the 3 has the PixelClock locked to specific values, which makes it difficult to generate timings with a calculator.

What we have done is from the values allowed in the pi3 to approach and reach the necessary timings, forcing the calculation a little by hand, although it does not match the pixelclock that we have to set.

Is this for autoswitching res in Recalbox for Pi3 or for another device?

digitalLumberjack, 2021-07-12:
Hi @mortaca,

It's for pi4 in fact. We are adding the support of dac -> scart in recalbox : 
https://gitlab.com/recalbox/recalbox/-/issues/1805

All your timings are running good, but I would love to test other ones to add some features to recalbox, with automatic drm mode selection from retroarch (zero config).

But i'm struggling with timing creations.

mortaca, 2021-07-12:
On the pi4 you can generate any modeline using the calculators available for PC, for example the CRT SwitchRes from calamity which is highly optimized for arcade monitors, then you just have to translate the modeline format to dpi_timings and it should work.

Do you already have the changes working with the new kernel to be able to make resolution changes on the fly?

digitalLumberjack, 2021-07-12:
Yeah everything is on the issue https://gitlab.com/recalbox/recalbox/-/issues/1805 and in the related Merge Request.

You are looking for the dpidac module.

rtomasa, 2021-07-12:
Cpasjuste sent me the updated bridge for kernel 5.10 2 weeks ago but we had no time to test yet. The old one for 5.4 did not fully work. Also you need to add the full timing database in both the txt file and inside the .dtbo to work

digitalLumberjack, 2021-07-12:
Yes @rtomasa, the last revision of the dpidac is working well.

We will have to provide the `timings.txt` with all possible modes (and for recalbox for all possible dac), and the configuration of the viewport have to be done on every emulator.

The limitation for now is that you can have only one mode for a given (resolution,framerate).

I'm checking right now how I can use the native resolution returned by the core to make it automatic with retroarch at least.

And to test that, i'm trying to make a 256*224p@60HZ timing or a 299*224p@60HZ to see how it could run on the screen.

Do you think it is possible, can I have such resolutions ? If it is valid resolutions, would you help me finding timings ?

If not, i'm modifying retorarch to add a new RATIO mode that will stretch the width to to 1920 but keep the core provided height.
That would do the work for many emulators.

rtomasa, 2021-07-12:
Mmm OK so to recap and refresh my mind, I was reading the latest changes on Cpasjuste kernel modules and I see that now it is not required to include all the timings inside the overlay files. Instead, a function called dpidac_get_modes will read all from /boot/timings.txt.

Assuming that, as per your comments, a timing is the combination of resoluton + refresh rate you cannot define two different timing for, let's say, 320x240@60
In any case, this should not be an issue since the modeline should be always the same for same combinations of resolution and refresh rate.

AFAIK there is no way to get a core timing from outside the retroarch. Anyhow, I'm in contact with both the creators of Retroarch's switchres and Calamity. Basically if I'm able to make this working as it is right now, they will integrate the autoswitching in retroarch and we could use Calamity's timing calculator to generate the full timings.txt database

rtomasa, 2021-07-28:
I was testing the latest changes yesterday and I found out that I don't get any audio in Pi4, neither from the GPIO nor the headphones jack. I've messed around with all config.txt parameters and nothing. Do you get any audio when the vga666 overlay is enabled?

rtomasa, 2021-07-28:
OK I found the issue. Audio is now working properly. What is not working is the 480i I need to check further if it's an issue with the timing parameters or is not supported for some other reason

digitalLumberjack, 2021-08-11:
I have audio yes :d

And for the record, the kernel must be patched for 480i mode.

rtomasa, 2021-08-14:
Question, did you finally submit any change into Retroarch mainline for the
automatic switchres/timings?
If so, are all timings working fine without adding the same into the .dtbo
file as required at the very beginning of the Cpasjuste development?

Thanks

digitalLumberjack, 2021-08-15:
Hi,

No timings still have to be listed for the rpidac module.
I almost got it working without it, but i'm stuck with switchres.

So retroarch selects the right mode at startup depending on core needs.

digitalLumberjack, 2021-08-16:
> Question, did you finally submit any change into Retroarch mainline for the
automatic switchres/timings?

Not yet, but the real objective here is to make the switchres lib work again on pi4, that i did not succeed yet.

rtomasa, 2021-08-16:
I’ve informed Alphanu and Subtring about the current status and they are
waiting for my alpha distro to test and start developing the new switchres
functionality using Calamity engine.