mortaca/RGB-Pi #66: Resolution changed every time when I quit from the emulator

wolfshow, 2021-08-18:
Anything I can do to avoid this?

rtomasa, 2021-08-18:
Resolution changes every time you enter and exit from any game.
Without further explanation I don't understand what's your issue?

wolfshow, 2021-08-18:
Thanks @rtomasa for the reply! I mean the resolution becomes weird with a very slim UI, but when I reboot, the resolution becomes normal.

wolfshow, 2021-08-18:
btw: I use the SONY PVM-9044QM monitor.

rtomasa, 2021-08-18:
The timing change is fully independent from the TV/Monitor used, so maybe you have some kind of corrupt file in the OS image, otherwise almost 2k people using the OS should be facing the same issue. Are you in the latest version?

wolfshow, 2021-08-18:
@rtomasa Yes, the latest version. I will check my timing change.

wolfshow, 2021-08-18:
https://youtube.com/shorts/jgNGWq_nmRE?feature=share

rtomasa, 2021-08-18:
It's keeping the super resolution from the game. As I mentioned, this is probably because the file containing the UI timings is corrupted, otherwise everyone should be facing the same issue. I suggest to write a new copy of the OS in another SD card to check

wolfshow, 2021-08-25:
@rtomasa just add one thing. I am using the retrotink ultimate board with rpi 3b+. Someone else told me that he had the same experience like I have, maybe because the rgb-pi does not support the retrotink ultimate board very well,  vga666 might be a better choice.

rtomasa, 2021-08-25:
This is the official git for rgb-pi hardware and software. We only provide support for rgb-pi cable and jamma board. If you have issues with other devices, you should make use of their support.

closed 2021-08-25 by rtomasa