mortaca/RGB-Pi #70: Key step missing from "Adding ROMs" Wiki section (no direction to "search games" to have system recognize the games)

Dochartaigh, 2022-03-02:
If you are no longer tracking tickets here hopefully somebody will let me know (last issue posted is from 5 months ago which is why I ask... also PLEASE let me know where to ask questions/post issues then: Discord AND Telegram are NOT useful... maybe 1 out 4 questions I post there gets an answer... and they're simple things...).

Anyway, just wanted to let you know you are missing a KEY step in the "Adding ROMSs" section of your Wiki. You never told people they have to select "Search Games" after they load their games in order for them to show up. Without this step people can not play a single game...

That isn't a logical option to choose either - on RetroArch for example 'search games'  will search all your game names for the one you just typed in... that's why I never thought to hit that option to LOAD these games into the playlists - as that just wasn't logical to me (or many other people).

[image]

emotionless-engine, 2022-03-05:
Someone listened and added another section at the bottom of page.

Thanks guys.

closed 2022-03-06 by rtomasa

Dochartaigh, 2022-03-06:
Yeah!!  Thank you to whoever did this. I already figured it out but hope it'll help others! (as others were lost with this who I saw post about the same issue on your Discord!).