mortaca/RGB-Pi #75: Screen rotation makes non mame roms vanish!

Steve772022, 2022-05-22:
Hi all 

Just to report an issue that's given me a migraine trying to figure out for over a week and cost me £50 because I thought I had a faulty RP so wanted to be sure. Basically I was unable to get any roms to appear after a games search other than mame, and not all mame roms at that. I went through every possible check to try and figure out what I was doing wrong, turns out nothing. 

If you  turn on screen rotation, no roms other than some mame ones will appear no matter what you do or what you put where. Turn screen rotation off and you don't even have to re search they just all re-appear again as if by magic. What a pain in the A%%E I'm just letting people know it's a thing so it doesn't drive anyone else mad wondering what they did wrong.  Hopefully someone can fix this issue, I know I can rotate the screen on a RP2 and RP3 in the config via display_rotate=3 etc ok fair enough. But on a RP4 not so easy and yes the same problem appears to exist on the OS4 too cos I tried it. So there hopefully that helps someone out.

rtomasa, 2022-05-22:
This is not a issue but a feature described in the official Wiki. It is very well known by all users. When rotating the UI, games are filtered so only TATE games are displayed.

Steve772022, 2022-05-22:
ok so where exactly does it say this in plain english in the tutorial section for this product, I'm sorry but nowhere does it say if you rotate the screen vertically not all roms will be displayed in a games search. I may have missed that but I certainly didn't see it on the main page and if it is there it's not obvious in my opinion because nobody I've asked about this online knows anything about it. And I've asked.
But hey never mind everyone must know because you just told me so ignore me

Thanks for the help, not sure why the person who emailed me a week ago couldn't have just said that to me instead of go to discord.

Anyway all the best.

rtomasa, 2022-05-22:
We only support people with original hardware and Discord is the official place to ask for any question.
The info is in https://www.mortaca.com/rgb-pi/wiki/index.php?title=Display_Options

closed 2022-05-22 by rtomasa