mortaca/RGB-Pi #76: Display > Trinitron Color Fix is missing

ElixiumReal, 2022-06-03:
Display > Trinitron Color Fix is missing from the menu in OS4 for Pi4/400. This is a requirement for me personally as I only have Trinitron TVs.

Is there some way I can activate it or do I need to buy a Pi2/3 and install "OS2 for Pi2/3"?

rtomasa, 2022-06-03:
This feature was already removed in the latests versions of the OS2 almost 2 years ago, so no matter what Pi or OS version you download. People using an old RGB-Pi cable + Affected trinitron TV were offered with a free upgrade/fix of their cables for a limited period of time, but that was long time ago as I said.

ElixiumReal, 2022-06-03:
Why would the feature be removed? Is there a special SCART cable that I need to buy?

ElixiumReal, 2022-06-03:
If I buy the new official RGB-Pi "SCART cable for CRT TV's" will this work with all Trinitron TVs? or is this still an issue?

rtomasa, 2022-06-03:
Because that fix was causing many other issues related to the timings. Also the fix is not feasible in OS4 since we use automatic/on the fly timing calculations. All "new" RGB-Pi devices include a hardware fix for the trinitron issue

rtomasa, 2022-06-03:
Talked to @mortaca and he says that you can still send your current cable for the upgrade if you like

ElixiumReal, 2022-06-03:
I haven't bought a cable yet, I've been using the RGBerry Pi hat with RGB Pi OS with my smaller (not Trinitron CRT, no color issue)

Just to confirm, If I buy a brand new RGB Pi cable from your ship, the trinitron issue has been fixed with the new cables?

rtomasa, 2022-06-03:
Yes

hellpe, 2023-03-13:
> This feature was already removed in the latests versions of the OS2 almost 2 years ago

Should the FAQ be updated then? [It still states](https://www.mortaca.com/rgb-pi/wiki/index.php?title=FAQ#I_have_a_Sony_Trinitron_TV_and_red_colors_look_muted.) that the Trinitron Color Fix menu option is available (instead of e.g. redirecting to [this page](https://www.mortaca.com/rgb-pi/wiki/index.php?title=Red_color_issue))

rtomasa, 2023-03-13:
Wiki updated

closed 2023-03-13 by rtomasa