mortaca/RGB-Pi #88: Excuse me, WHERE IS THE SOURCE CODE?

Keyvantajfar, 2024-02-27:
Hello. Hope this message finds you well.
I need more documentation and stuff on your work. But you have nothing.
GNU/LINUX is an open source and free software operating system. Raspbian/RetroPie/Debian or whatever else you've built your OS on top of it. That Forces your software to be Open source as well; just how the License says. Hope you answer me back. I'm not looking for any troubles. Sincerely,
Me.

rtomasa, 2024-02-27:
You are totally wrong in every single statement.
First of all there are many kinds of licenses and packages in the distribution.
Second, you did not specify what do you want, a package, kernel...? We use all standard Debian packages so nothing to provide here.
Third, every single custom piece of software we use are from different separated open source projects. If you want me to create a tech guide for you I must say sorry but no.
Fourth, RGB-Pi OS is a friend project but does not belong to RGB-Pi which is only the hardware. It is created by me, a single user which participates in the RGB-Pi community.

closed 2024-02-27 by rtomasa

Keyvantajfar, 2024-03-30:
> You are totally wrong in every single statement. First of all there are many kinds of licenses and packages in the distribution. Second, you did not specify what do you want, a package, kernel...? [...]

You're claiming that "Third, every single custom piece of software we use are from different separated open source projects.". Do you know that Open Source does not mean you can use it in your software and keep it closed sourced??

Most of the software you're using most probably have GPL3 license which forces you to OpenSource the modified version of the sources. We need to talk buddy. this is a super friendly request from me. im a geek and an advocate of FOSS community and i am responsible for this. i hope you get what i'm saying.

Keyvantajfar, 2024-03-30:
> You are totally wrong in every single statement. First of all there are many kinds of licenses and packages in the distribution. Second, you did not specify what do you want, a package, kernel...? [...]

Thanks for your response. Yeah i'm new here and i wanned to see how you've put this together. i mean i see no documentation nowhere so i'm lost! yeah like how your UI works and it is integrated to this OS and some information like that. i hope you can stay in touch with me cause i'd like to contribute to this project!